n0bel/PiClock3 #1: Having issue with Weather maps Displaying

taftr1, 2022-01-17:
I have edited the API Keys file and inserted the required API keys.  However when I run the clock it loads the clock page, but the forecasts and maps are blank (no map, no radar).  I've looked in the Logs but see no errors.  I have included the log file here.  I cant seem to locate in the Code where the Radar maps are processed.  I'm not an expert Python programmer
[PyQtPiClock3.log](https://github.com/n0bel/PiClock3/files/7878615/PyQtPiClock3.log)

n0bel, 2022-01-17:
PiClock3 is only partially complete.   The framework is done, but only the clock (Analog, Digital plugins), sunrise/sunset/moonphase (Astral plugins), date at the top (Date plugin) and current conditions (METAR plugin) modules are complete.   It is working for you as expected.   I was last working on radar on my local commits, but there were some major bugs so I didn't want to push that yet.   The plugin section of the Config will show you what's enabled.   No Radar, No Forecast plugins are activated (nor do those folders exists in the source tree).

n0bel, 2022-01-17:
I've updated the Readme page of PiClock3 to reflect what is complete.

taftr1, 2022-01-17:
Thanks....  Yes.  Everything you mentioned is working well....  I'll monitor here for updates.   Thanks for all the hard work

closed 2026-08-25 by n0bel