n0bel/PiClock3 #4: Metar data not updating

lupis1111, 2022-03-12:
I’ve installed PiClock3 on a couple of different Pi’s per the directions in the readme file and I’m finding that the Metar data is not updating on either one.  Is that expected with the current state of the program.

lupis1111, 2023-06-05:
Thanks @mtijrobinson, that fixed my issue.  Looking forward to updates to PiClock3, it’s a great program.

SerBrynden, 2023-06-05:
Have you tried this fork of the original PiClock? https://github.com/SerBrynden/PiClock
It has more commits than PiClock3.  It's been updated to Python3 and PyQt5 to run on the latest Raspberry Pi OS and has everything working, such as radar and forecasts.  n0bel and I have been working on the original PiClock more so than PiClock3.

mtijrobinson, 2023-06-05:
Awesome! I'll check that out. I like the modular plugin design of what he has going on here.

SerBrynden, 2023-06-05:
Yeah, it's the next generation of PiClock, but hasn't seen much activity in awhile.  And I think n0bel has been absent for a few weeks too.

closed 2026-08-25 by n0bel