n0bel/PiClock3 #5: Unhandled Error - AttributeError: 'Config' object has no attribute 'styles'

Rabidgeek, 2022-04-21:
I'm able to run, customize a bit and run again PiClock3 manually. No problems there. However, trying to get it to start on reboot - tried two ways, first was a crontab -e, added '@reboot /usr/bin/python3 /home/pi/PiClock3/PyQtPiClock3.py' but it refuses to run because qt.qpa.plugin and qt.qpa.xcb could not connect to display, need to reinstall Qt... so I wrote a display.desktop in /etc.xdg/autostart with similar command to run PiClock3. However, it gives me the Unhandled Error on boot and does not execute the program. I can open terminal and run python3 PyQtPiClock3.py and it starts up just fine! Seems problem is running this so early in startup?

looking forward to your map update!

Rabidgeek, 2022-04-21:
Thanks - I did that, and now the logs are back to Qt errors. this one says:
"Invalid MIT-MAGIC-COOKIE-1 keyqt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found. 
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem. 

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted"

Even though on reboot this happens, I just have to open terminal and run python3 PyQtPiClock3.py and all is fine.

SerBrynden, 2022-04-21:
Beats me.  Maybe n0bel or others will have some ideas.

Rabidgeek, 2022-04-23:
Thanks for your help, though. If its of any help, I'm running a Raspberry Pi 4, 4gb ram, all required packages installed to the most recent version, with a small samsung tv. Tv was on the entire time during reboots. So far I've been running the PiClock3 since my last reply without issue. Been working on customizing the fonts, but struggling with figuring out how to apply shadows. I've reinstalled pyqt5 twice, too. Maybe n0bel's next update will fix my issue? fingers crossed!

n0bel, 2026-08-25:
Fixed in `055cb85` — `initData()` now guards the read instead of assuming a `styles:` block exists.

It's better than a guard now, in fact: styling moved into themes, so none of the shipped example configs has a `styles:` block at all any more. Every run takes the path that used to crash, which means it can't silently regress.

On the Qt errors further down the thread — that's a different problem, and not one PiClock3 causes. `Invalid MIT-MAGIC-COOKIE-1 key` is X refusing the connection on authorisation: the process found the display but had no valid cookie for it. That's what `@reboot` cron and some autostart paths do — they give you `DISPLAY` but not `XAUTHORITY`. Running it from a terminal works because that shell already has both from your desktop session.

Worth noting `startup.sh` is a PiClock v1 file. PiClock3 has never shipped one and has no autostart tooling of its own, so there's nothing here for me to fix. v1's script sets `DISPLAY` but not `XAUTHORITY`, which is likely why adopting it moved you from one error to the other.

Closing this as the reported bug is fixed. If autostart is still biting you four years on, open a fresh issue and we can work out what your session needs.

closed 2026-08-25 by n0bel