n0bel/PiClock3 #9: SyntaxError: Missing parentheses in call to 'print'. Did you mean print("map base url: " + self.baseurl)?

Angrenost1, 2024-09-19:
Hello n0bel,
First, I think you PiClock really rocks.
I am trying to install it on my Raspberry Pi model 4B with Raspbian Bullseye installed (the Pi would not start up with Stretch).
I followed all command, but when I try to start the PiClock with the command "sh start.sh -n -s" I receive an error below.
I really hope you can help me to solve.
Regards
Remco

Error
--------------------------------------------------------------------------------------------
root@PiClock:~/PiClock# sh startup.sh -n -s
Disabling screen blanking....
No protocol specified
xset:  unable to open display ":0"
No protocol specified
xset:  unable to open display ":0"
No protocol specified
xset:  unable to open display ":0"
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Starting NeoPixel Ambilight Service...
Checking for GPIO Buttons...
Starting gpio-keys Service...
Checking for Temperature Sensors...
gpio memmap: Operation not permitted
Starting PiClock.... logging to screen.
  File "/root/PiClock/Clock/PyQtPiClock.py", line 1642
    print "map base url: " + self.baseurl
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("map base url: " + self.baseurl)?

BertLindeman, 2024-09-19:
Hello,

The error you see is a python3 message.
Python3 requires parens where python2 did not.

I am not current with my knowledge about other PiClock repositories.

I see 3 options
1. edit this python script so all print commands have parentheses.
2. install python2 and use it for the PiClock
3. find another repository forked from this one that **is** python3 ready

Long time ago I did the first. No rocket science, but nowadays there might be another repo more current.

Ah, found one see [SerBrynden repositories](https://github.com/SerBrynden)

Bert

Angrenost1, 2024-09-20:
@BertLindeman Thanks for your quick reply. Will have a look at the link you send.

Angrenost1, 2024-09-22:
@BertLindeman I used the for of Ser Brynden and I confirm that it is working brilliantly. I now only have to figure out how to reduce the speed of the weather radar. Again thanks for your suggestion!

BertLindeman, 2024-09-22:
which repository do you use? I assume PiClock (not PyClock3 as that is not yet ready).
In [PyQtPiClock.py](https://github.com/SerBrynden/PiClock/blob/master/Clock/PyQtPiClock.py) 

I see:
```python

try:
    Config.radar_refresh
except AttributeError:
    Config.radar_refresh = 10  # minutes

```
So there is in the config:

```python

metric = 0  # 0 = English, 1 = Metric
radar_refresh = 10  # minutes
weather_refresh = 30  # minutes

```

That should get you off ;-)

Bert