n0bel/PiClock3 #10: No module named 'yamlinclude'

alexsalex, 2024-10-10:
```
pi@PyClock:~/PiClock3 $ python3 ./PyQtPiClock3.py
Traceback (most recent call last):
  File "/home/pi/PiClock3/./PyQtPiClock3.py", line 6, in <module>
    from PiClock3.PiClock3 import PiClock3
  File "/home/pi/PiClock3/PiClock3/PiClock3.py", line 5, in <module>
    from yamlinclude import YamlIncludeConstructor
ModuleNotFoundError: No module named 'yamlinclude'
```

SerBrynden, 2024-10-15:
Did you install all the required packages according to the readme? -> https://github.com/n0bel/PiClock3#:~:text=No%20detailed%20instructions%20have%20been%20created.%20Basic%20information%20follows.

gamename, 2024-12-08:
@SerBrynden 
I installed the pre-reqs and got the same error.

SerBrynden, 2024-12-08:
PiClock3 isn't finished anyway.  Try using https://github.com/SerBrynden/PiClock

gamename, 2024-12-08:
> PiClock3 isn't finished anyway. Try using https://github.com/SerBrynden/PiClock

Thanks!